colinrgodsey/step-daemon #1: Please explain ticks-per-second value in configuration

pandel, 2017-08-25:
Hi!

First, your project sounds very interesting and I'd like to play around with it a bit.

A question regarding the configuration file: what is the ticks-per-second value for and how do I determine the correct one for my printer? All the other values are well known, but I cannot figure out what ticks-per-second is good for.

Cheers,
Holger

colinrgodsey, 2017-09-12:
Hey! Sorry, I've been traveling for a few weeks. Very excited to see some interest in the project, can't wait to see what horrors you find ;)

So ticks-per-second is definitely the most unique setting here. Basically with the new Marlin extension, the chunked "step buffers" that come in can specify a tick-rate to be used for processing that data. Eventually this is something you can specify in g-code, but for the most part you just want to configure it in the config and keep it constant. 

Depending on what features you have enabled in Marlin, the UART serial speed, and the Marlin device clock rate, there will be an ideal max tick rate that works best for your setup. Basically the tickrate sets the actual stepper interrupt rate internally in the marlin device. This rate specifies how fast the chunk data is consumed, sets acceleration accuracy, and ultimately sets the max tick rate you can run your steppers at (technically the stepper can run at 7/8's of this tick rate). With lin_advanced disabled on a 16MHz Marlin device, I've been pretty happy with 32k ticks/s. 

(lin_advanced feature coming soon to step daemon)

pandel, 2017-09-21:
Hey Colin! Nothing to feel sorry for! I'm a somewhat busy, too, at the moment... Hope you had a nice trip ;-)

Thanks for your explanation! When I have time, I try to take a closer look. At the moment I am fighting with my board and some bilinear leveling issues I don't understand (whole bed "tilted" to left side after leveling is applied). Infact, I had to go back to linear to get it working...

But regardless of that, I'm interested in how your project evolves because of the limited speed and prog memory of my printer control board. If it would be possible to overcome the hardware limitations using your daemon, it could be a perfect addition to my octoprint setup :)

closed 2017-10-23 by pandel