colinrgodsey/step-daemon #3: Dual Z axis - and a guide

dmalyavin, 2020-05-27:
Hi, Will this work witha dual z axis setup? I cant see any SP_5x2_256 config or something like that ... also would be great to get a step by step guide on how to set this up on an existing Octopi, I am able to clone the repository, but missing some infor on how to update part of the config file, and also how to compile and run. If you have a bit of time to show me I can write up a guide for people. Thank you!

colinrgodsey, 2020-05-28:
Great to see some interest in the octoprint plugin! I'm still finishing some things up there (docs, better status reporting and the distributable), should hopefully have the basic stuff done today. 

As far as the dual-z, that should still be handled by marlin with the single z axis part of the direct stepping data (it'll still do the identical pulses for up to 4 z steppers). Homing for multi-z should still work normal.

colinrgodsey, 2020-05-28:
And I'd also definitely appreciate any help with getting the docs in order. Once I get some of the basic stuff done, I would be hugely interested in getting any feedback on next-step improvements (workflow and docs)!

colinrgodsey, 2020-05-29:
@dmalyavin alright, got a good beta release of the plugin finally: https://raw.githubusercontent.com/colinrgodsey/maven/master/step-daemon/octoprint-plugin/latest.zip

It has the documentation embedded in it now, should be a little easier to get up and going quick

closed 2020-05-29 by colinrgodsey