colinrgodsey/step-daemon #4: SBT Error

dmalyavin, 2020-06-03:
Hello when trying to run sbt fro cloned repository I get the following error:

The java installation you have is not up to date
requires at least version 1.6+, you have
version 11.0.7

pi@octopi:~/ss/step-daemon $ javac --version
javac 11.0.7

Dont have anything special on this Octopi, just a few plugin.

colinrgodsey, 2020-06-04:
Interesting, that is definitely a bit weird... did you use the bundled SBT jar in the repo, or SBT installed on the system level?

colinrgodsey, 2020-06-06:
Might also be able to try the octoprint plugin directly. Should use the bunbled SBT. havent tried in on octopi yet, but works fine for raspbian

CRCinAU, 2020-07-09:
I get the same issue...

```
$ git clone https://github.com/colinrgodsey/step-daemon.git
$ cd step-daemon
$ ./sbt clean server/assembly
 
The java installation you have is not up to date
requires at least version 1.6+, you have
version 11.0.7

Please go to http://www.java.com/getjava/ and download
a valid Java Runtime and install before running .

$ javac --version
javac 11.0.7
```

CRCinAU, 2020-07-09:
Editing the file `sbt-files/sbt-launch-lib.bash` to change the following works around the problem:

```
# Detect that we have java installed.
checkJava() {
  return

  local required_version="$1"
  # Now check to see if it's a good enough version
  if [[ "$java_version" == "" ]]; then
    echo
    echo No java installations was detected.
    echo Please go to http://www.java.com/getjava/ and download
    echo
    exit 1
  elif [[ ! "$java_version" > "$required_version" ]]; then
    echo
    echo The java installation you have is not up to date
    echo $script_name requires at least version $required_version+, you have
    echo version $java_version
    echo
    echo Please go to http://www.java.com/getjava/ and download
    echo a valid Java Runtime and install before running $script_name.
    echo
    exit 1
  fi
}
```

CRCinAU, 2020-07-09:
Also fixed in SBT here: https://github.com/sbt/sbt-native-packager/pull/1111/commits/28416776f62d7457cf90f93f71a7479a54430923

colinrgodsey, 2020-12-19:
go everything rewritten in go no, much better build management

closed 2020-12-19 by colinrgodsey