colinrgodsey/step-daemon #5: "Could not open port /tmp/pty-stepd-client"

Toby406, 2020-06-15:
Hi,
I tried to run your plugin on a ubuntu-server laptop with two cores.
After installing the Octoprint plugin with it's dependencies, changing the baudrate and restarting I get the following error in the Teminal:
Unexpected error while connecting to serial port: AUTO SerialException: '[Errno 2] could not open port /tmp/pty-stepd-client: [Errno 2] No such file or directory: '/tmp/pty-stepd-client'' @ comm.py:_openSerial:2691 (hook stepd)

colinrgodsey, 2020-06-16:
The workflow is still a bit weird for the startup/install portion of the server software for the plugin.  If you click on the stepd 'tab' in the octoprint UI, you'll be able to see the status of the install/startup. The first update will take several minutes, and it should show 'updating' in the stepd tab. If the update does seem hung, you can check the octoprint log for any other further build or whatnot related errors.

Toby406, 2020-06-16:
After being a bit more patient, I managed to connect. But I can neither start a print or read the current temperature. I think I will wait until this really cool project reaches a more "finished" state, I love the idea and I will watch your career with great interest.

colinrgodsey, 2020-12-19:
the go rewrite has much better serial connection management, should be less of this

closed 2020-12-19 by colinrgodsey