colinrgodsey/step-daemon #7: OctoPrint Plugin Not Starting stepd

pearson, 2020-07-10:
I'm just starting out with step-daemon, so I could easily be missing something here... :-) 

Short story: after uninstalling and reinstalling the OctoPrint plugin, stepd is not started automatically.

Setup:
- OctoPrint runs on a Raspberry Pi 3B+. The Pi is running the latest Raspberry Pi OS (32 bit) and is updated.
- OctoPrint is running v1.4.0 stable. It does have a couple 3rd party plugins.

The steps I had taken:
1. I initially installed the OctoPrint plugin using the URL (it did not show up on the Plugin Manager's list.) This appears to have started stepd -- stepd showed up on the process list.
2. I uninstalled the OctoPrint plugin using the "clean up" option. 
3. The process list showed stepd still running. I rebooted the Raspberry Pi.
3. I reinstalled the OctoPrint plugin. However, stepd was not started. 
4. Restarting OctoPrint, and then restarting the Pi, did not start stepd.

I had not cloned the server code, and the only install I had done on the Raspberry Pi had been the plugin.

(The plugin uninstall and re-install were due to not getting the communication with the printer working. That's a different issue for me to sort out. :-) )

colinrgodsey, 2020-07-20:
I'm currently rewriting this project in go. Automating the build/run from octoprint using the current scala codebase is way too brittle and messy. hoping to have the rewrite done soon, should make the octoprint plugin far more stable

colinrgodsey, 2020-12-19:
rewrite is done now, should be much more direct connection management

closed 2020-12-19 by colinrgodsey