colinrgodsey/step-daemon #10: Connection Error

rafaelbiasi, 2020-10-08:
When I try to connect to the printer the error happens:

Offline (Error: TypeError: 'write() argument must be str, not bytes' @ comm.py:_do_send_without_checksum:3593)
[image]

[octoprint.log](https://github.com/colinrgodsey/step-daemon/files/5350747/octoprint.log)

rafaelbiasi, 2020-10-10:
I have almost no Python skills.

I replaced the 'subprocess' with 'pexpect' and managed to make it work. I also comment StepdServiceLogger.
Then, I increase 'devStettingsTimeout' in 'pipeline/config_handler.go', because my printer have a slow initialization.

So, now, I got the following error in terminal tab:
Recv: M420V1                                                                                                                                              │
Recv: error: failed parsing gcode (strconv.Atoi: parsing "420V1": invalid syntax)

Then loss of connection to printer.

rafaelbiasi, 2020-10-10:
I got work.
This 'M420V1' gcode is sent by ABL Expert Plugin. I edited to send with space: 'M420 V1'

colinrgodsey, 2020-12-17:
good catch! there should definitely be more graceful handling for general parsing errors like that

Kannix2005, 2020-12-31:
Hey, i got the same problem, can you describe the steps you´ve done to fix this?
I honestly got no idea.

Best

colinrgodsey, 2021-01-18:
@Kannix2005 what g-code is it failing on? you should be able to find that g-code (or the source of it) and modify it to correct the gcode.

colinrgodsey, 2021-01-18:
This should be fixed in the latest commit. If you're using OctoPrint, you'll need to grab the new plugin (URL listed on the README) as the new version requires the newer plugin. Let me know if its still causing issues!

closed 2021-01-18 by colinrgodsey