colinrgodsey/step-daemon #13: Marlin Compatibility

drphil3d, 2020-12-04:
Do I have to use the Direct Step branch of marlin? is that the only branch that will work?

```
Changing monitoring state from "Offline" to "Opening serial connection"
Connecting to Step Daemon
Step Daemon is still updating.
Changing monitoring state from "Opening serial connection" to "Error: Connection error, see Terminal tab"
Changing monitoring state from "Error: Connection error, see Terminal tab" to "Offline (Error: Connection error, see Terminal tab)"
Unexpected error while connecting to serial port /dev/ttyUSB0, baudrate 115200 from hook stepd: Exception: 'Step Daemon is still updating.' @ comm.py:_open_serial:3670
Changing monitoring state from "Offline" to "Opening serial connection"
Connecting to Step Daemon
Changing monitoring state from "Opening serial connection" to "Connecting"
Connected to: <octoprint_stepd.StepdService.StepdService instance at 0x6119a328>, starting monitor
Send: N0 M110 N0*125
Recv: warn:failed to load ./bedlevel.json: open ./bedlevel.json: no such file or directory
Send: N0 M110 N0*125
Recv: info:config processed
No answer from the printer within the connection timeout, trying another hello
There was a timeout while trying to connect to the printer
Changing monitoring state from "Connecting" to "Offline (Error: )"
Connection closed, closing down monitor
Changing monitoring state from "Offline" to "Opening serial connection"
Connecting to Step Daemon
Changing monitoring state from "Opening serial connection" to "Connecting"
Connected to: <octoprint_stepd.StepdService.StepdService instance at 0x6c83f1c0>, starting monitor
Send: N0 M110 N0*125
Recv: warn:failed to load ./bedlevel.json: open ./bedlevel.json: no such file or directory
Recv: info:config processed
Send: N0 M110 N0*125
Unexpected error while reading serial port, please consult octoprint.log for details: ValueError: 'I/O operation on closed file' @ comm.py:_readline:3831
Changing monitoring state from "Connecting" to "Offline (Error: ValueError: 'I/O operation on closed file' @ comm.py:_readline:3831)"
Connection closed, closing down monitor
Changing monitoring state from "Offline" to "Opening serial connection"
Connecting to Step Daemon
Changing monitoring state from "Opening serial connection" to "Connecting"
Connected to: <octoprint_stepd.StepdService.StepdService instance at 0x6c464490>, starting monitor
Recv: warn:failed to load ./bedlevel.json: open ./bedlevel.json: no such file or directory
Send: N0 M110 N0*125
Recv: info:config processed
No answer from the printer within the connection timeout, trying another hello
Send: N0 M110 N0*125
There was a timeout while trying to connect to the printer
Changing monitoring state from "Connecting" to "Offline (Error: )"
Connection closed, closing down monitor
```

colinrgodsey, 2020-12-17:
Sorry, i just realized the README still points to the dev branch. Any new 2.0.x version is fine. The README does need some additional cleanup.

Ex0ry, 2020-12-19:
I got the exact same error on Marlin 2.0.7.2 and 2.0.7.
It seems like not every version is compatible.

Does anyone know wich version to use or how to fix this?

colinrgodsey, 2020-12-19:
Hmm very interesting indeed. The version im running now is: https://github.com/MarlinFirmware/Marlin/commit/131ddeaeba96addcf573cbeff86b65225d3b8320

It'd definitely be interesting to see if the problem persists on that version. If it does, then there might be some other config interaction at work here.

Ex0ry, 2020-12-20:
Just installed the marlin version you have provided, but can't get it to work..
Keep getting those two errors:
```
Changing monitoring state from "Offline" to "Opening serial connection"
Connecting to Step Daemon
Changing monitoring state from "Opening serial connection" to "Connecting"
Connected to: <octoprint_stepd.StepdService.StepdService instance at 0x6857d558>, starting monitor
Send: N0 M110 N0*125
Recv: warn:failed to load ./bedlevel.json: open ./bedlevel.json: no such file or directory
Recv: info:config processed
Send: N0 M110 N0*125
Unexpected error while reading serial port, please consult octoprint.log for details: ValueError: 'I/O operation on closed file' @ comm.py:_readline:3831
Changing monitoring state from "Connecting" to "Offline (Error: ValueError: 'I/O operation on closed file' @ comm.py:_readline:3831)"
Connection closed, closing down monitor
Changing monitoring state from "Offline" to "Opening serial connection"
Connecting to Step Daemon
Changing monitoring state from "Opening serial connection" to "Connecting"
Connected to: <octoprint_stepd.StepdService.StepdService instance at 0x69294cd8>, starting monitor
Send: N0 M110 N0*125
Recv: warn:failed to load ./bedlevel.json: open ./bedlevel.json: no such file or directory
Send: N0 M110 N0*125
Recv: info:config processed
No answer from the printer within the connection timeout, trying another hello
Send: N0 M110 N0*125
There was a timeout while trying to connect to the printer
Unexpected error while writing to serial port: IOError: '[Errno 32] Broken pipe' @ comm.py:_do_send_without_checksum:4686
Changing monitoring state from "Connecting" to "Offline (Error: IOError: '[Errno 32] Broken pipe' @ comm.py:_do_send_without_checksum:4686)"
Connection closed, closing down monitor
Changing monitoring state from "Offline" to "Opening serial connection"
Connecting to Step Daemon
Changing monitoring state from "Opening serial connection" to "Connecting"
Connected to: <octoprint_stepd.StepdService.StepdService instance at 0x69282198>, starting monitor
Send: N0 M110 N0*125
Recv: warn:failed to load ./bedlevel.json: open ./bedlevel.json: no such file or directory
Send: N0 M110 N0*125
Recv: info:config processed
Unexpected error while reading serial port, please consult octoprint.log for details: ValueError: 'I/O operation on closed file' @ comm.py:_readline:3831
Changing monitoring state from "Connecting" to "Offline (Error: ValueError: 'I/O operation on closed file' @ comm.py:_readline:3831)"
Connection closed, closing down monitor
```

Can you provide a config that is known to work?

jcherniak, 2021-01-12:
I'm hitting the same thing.  Any idea what might be up?  I'm on an Ender 3 Pro

colinrgodsey, 2021-01-18:
Sorry for the delay, finally got around to updates, including fixed support for python2/3 octoprint. This has been tested with the latest 2.0.x version of Marlin. 

You should be able to grab the new OctoPrint plugin here: https://raw.githubusercontent.com/colinrgodsey/maven/master/step-daemon/octoprint-plugin/latest.zip and you should be able to install it over the existing plugin and reset.

Let me know if there's still issues!

closed 2021-01-18 by colinrgodsey