colinrgodsey/step-daemon #14: '[Errno 32] Broken pipe' Error on latest Marlin build

Ex0ry, 2020-12-19:
Hello, 

Previously i had marlin 2.0.7 installed and got the same error output as on issue #13.
Thinking that a marlin update would solve the issue (is that the conclusion?), i went to update marlin to 2.0.7.2.

However when trying to connect now, i always get the error pasted below.
Do i have to use a specific version of marlin?

`Changing monitoring state from "Offline" to "Detecting serial connection"
Performing autodetection with 1 port/baudrate candidates: /dev/ttyACM0@250000
Trying port /dev/ttyACM0, baudrate 250000
Connecting to Step Daemon
Handshake attempt #1 with timeout 2.0s
Connected to: <octoprint_stepd.StepdService.StepdService instance at 0x677ca558>, starting monitor
Recv: warn:failed to load ./bedlevel.json: open ./bedlevel.json: no such file or directory
Send: N0 M110 N0*125
Recv: info:config processed
Handshake attempt #2 with timeout 2.0s
Unexpected error while writing to serial port: IOError: '[Errno 32] Broken pipe' @ comm.py:_do_send_without_checksum:4686
Changing monitoring state from "Detecting serial connection" to "Offline (Error: IOError: '[Errno 32] Broken pipe' @ comm.py:_do_send_without_checksum:4686)"
Connection closed, closing down monitor
Closing down send loop`

Ex0ry, 2020-12-19:
I dont know if that plays a role, but when SERIAL_OVERRUN_PROTECTION is disabled i get the error mentioned above.
But when it is enabled i get the exact same error as in issue #13.

colinrgodsey, 2021-01-18:
Sorry for the delay, finally got around to updates- including 

You should be able to grab the new OctoPrint plugin here: https://raw.githubusercontent.com/colinrgodsey/maven/master/step-daemon/octoprint-plugin/latest.zip

You should be able to install it over the existing plugin and reset. 

If you are still having issues, please double check that the baudrate you're using to connect in OctoPrint matches the baudrate on the marlin build. If it's still timing out, please attach any related logs from octoprint (there's additional messaging there that's not exposed through the console).

colinrgodsey, 2021-01-18:
Closing this out for now, please re-open if the issue somehow persists

closed 2021-01-18 by colinrgodsey