colinrgodsey/step-daemon #15: Error no connection to printer

Seelenkind, 2021-02-03:
whats wrong here, what have to do ?

Changing monitoring state from "Opening serial connection" to "Error: Connection error, see Terminal tab"
Changing monitoring state from "Error: Connection error, see Terminal tab" to "Offline (Error: Connection error, see Terminal tab)"
Unexpected error while connecting to serial port /dev/ttyACM0, baudrate 115200 from hook stepd: Exception: 'Step Daemon is still updating.' @ comm.py:_open_serial:3670

tiagofreire-pt, 2021-02-27:
Same issue here:

```
Connecting to Step Daemon
Unexpected error while connecting to serial port /dev/ttyUSB0, baudrate 115200 from hook stepd: FileNotFoundError: '[Errno 2] No such file or directory: '/home/pi/.octoprint/data/stepd/stepd': '/home/pi/.octoprint/data/stepd/stepd'' @ comm.py:_open_serial:3670
Could not open port /dev/ttyUSB0, baudrate 115200, skipping
Changing monitoring state from "Detecting serial connection" to "Error: No more candidates to test, and no working port/baudrate combination detected."
Changing monitoring state from "Error: No more candidates to test, and no working port/baudrate combination detected." to "Offline (Error: No more candidates to test, and no working port/baudrate combination detected.)"
```

Psyobilin, 2021-03-02:
Here too!
`Changing monitoring state from "Offline" to "Opening serial connection"
Connecting to Step Daemon
Changing monitoring state from "Opening serial connection" to "Error: Connection error, see Terminal tab"
Changing monitoring state from "Error: Connection error, see Terminal tab" to "Offline (Error: Connection error, see Terminal tab)"
Unexpected error while connecting to serial port /dev/ttyACM0, baudrate 250000 from hook stepd: FileNotFoundError: '[Errno 2] No such file or directory: '/home/openhabian/.octoprint/data/stepd/stepd': '/home/openhabian/.octoprint/data/stepd/stepd'' @ comm.py:_open_serial:3670`

Psyobilin, 2021-03-02:
I deleted my .octoprint folder and gave it a fresh start. that solved the problem for me.

bwiedmann, 2021-03-16:
`/home/pi/.octoprint/data/stepd/stepd` seems to be missing.

Is golang installed on your Pi? 

You can install it with

`sudo apt install golang`

cosworth99, 2022-01-12:
Did a fresh install using pi imager. Installed Golang. Ran the zip file in the plugin installer. Did this twice. Still getting a ValueError: 'I/O operation on closed file' @ comm.py:_readline:3940 error as well.

Specifically:

Unexpected error while reading serial port, please consult octoprint.log for details: ValueError: 'I/O operation on closed file' @ comm.py:_readline:3940
Changing monitoring state from "Operational" to "Offline after error"
Connection closed, closing down monitor

It will connect, handshake/echo a reply from an M503 then pop out of connection.

Once I turn the StepDaemon plugin off, everything is fine.