colinrgodsey/step-daemon #17: Unknown command when sending G1.

ChadDevOps, 2021-03-25:
I am attempting to run StepD with the octoprint plugin. Everything has started, but when sending any movement commands I get the following: Any ideas?

Example: `G1 X0 Y0 Z5 F4000`

```
Recv: echo:Unknown command: "!\x0f)"
Recv: echo:Unknown command: "!\x0e"
Recv: warn:pending OK count dropped below 0
Recv: echo:Unknown command: "!"
Recv: warn:pending OK count dropped below 0
Recv: echo:Unknown command: ""
Recv: warn:pending OK count dropped below 0
Recv: echo:Unknown command: "!\x0c"
Recv: warn:pending OK count dropped below 0
Recv: echo:Unknown command: "!\x0b"
Recv: warn:pending OK count dropped below 0
Recv: echo:Unknown command: "!"
Recv: warn:pending OK count dropped below 0
Recv: echo:Unknown command: ""
Recv: warn:pending OK count dropped below 0
Recv: echo:Unknown command: "!	"
Recv: warn:pending OK count dropped below 0
Recv: echo:Unknown command: ""
Recv: warn:pending OK count dropped below 0
Recv: echo:Unknown command: "!\x07"
Recv: warn:pending OK count dropped below 0
Recv: echo:Unknown command: "!\x06"
Recv: warn:pending OK count dropped below 0
Recv: echo:Unknown command: "!\x05"
Recv: warn:pending OK count dropped below 0
Recv: echo:Unknown command: "!\x04"
Recv: warn:pending OK count dropped below 0
Recv: echo:Unknown command: "!\x03"
Recv: warn:pending OK count dropped below 0
Recv: echo:Unknown command: "!\x02"
Recv: warn:pending OK count dropped below 0
Recv: echo:Unknown command: "!\x01"
Recv: warn:pending OK count dropped below 0
Recv: echo:Unknown command: "!"
Recv: warn:pending OK count dropped below 0
```

colinrgodsey, 2021-03-29:
@ChadDevOps are you using a 32-bit Marlin board? Direct Stepping in Marlin only supports 8-bit AVR boards at the moment. Support for the 32-bit ARM boards should be coming up soon- I have a functional version, but theres a USB Serial issue that I need to figure out first. I was close, but I've recently come into some housing complexities and will not have access to everything I need to test for another 2-3 months :(

ChadDevOps, 2021-03-29:
@colinrgodsey Ah! Yes, I have the 4.2.2 board in the Ender 3 Max. That would explain it :) Thanks for the follow-up.

Eliminateur, 2021-04-13:
> @ChadDevOps are you using a 32-bit Marlin board? Direct Stepping in Marlin only supports 8-bit AVR boards at the moment. Support for the 32-bit ARM boards should be coming up soon- I have a [...]

¿can you updated the readme to reflect this?, because nowhere it says that stepd only works with 8-bit boards until i read this issue, i was trying to make it work on my 32-bit board and now will stop after reading this.

Temtaime, 2021-12-11:
@colinrgodsey thanks for a good project. But is it dead ? No updated from January ...

colinrgodsey, 2021-12-11:
@Temtaime not dead, just big changes in personal situation. the next step i have to do is build a rig to test my smoothieboard (which is god knows where in storage) for 32b integration. it is mostly done, but i need to proceed with some real word tests (and possible deal with wonky data collection patterns for USB).

as far as the original target, the 8-bit boards, things should still be working fine.

WestwardWinds, 2022-06-10:
@colinrgodsey Hello, first thank you so much for your work on this, you are the only person working on a step daemon for Marlin, even now years later. Do you anticipate still working on 32 bit support for this? I just upgraded my Ender 3 Pro with a BTT SKR Mini E3 V3 which is 32 bit and would love to use direct stepping with it