colinrgodsey/step-daemon #19: Plugin not present on octoprint repo

Eliminateur, 2021-04-11:
The plugin is not present on the octoprint plugin repository, installing it manually from the URL worked perfectly

colinrgodsey, 2021-06-01:
I haven't submitted it to the octoprint reposity yet, as I wanted to get support for 32-bit printers in Marlin before that.

Hmm, as far as the plugin failing, does anything show up in the octoprint logs? There might be an issue with the golang complicate. Was golang installed on the system?

mytechguyri, 2021-07-23:
Doesn't work for me either.... Creality 4.2.7 board, THD3D unified FW v2 Marlin... I don't know however if I have everything configured correctly.   I see reference to "Copy example config from config.conf.example to config.conf" but there is no config.conf.example file anywhere that I can find, so who knows?

Here's what I'm seeing in the terminal window

Changing monitoring state from "Offline" to "Opening serial connection"
Connecting to Step Daemon
Changing monitoring state from "Opening serial connection" to "Connecting"
Connected to: <octoprint_stepd.StepdService.StepdService object at 0x6656c5b0>, starting monitor
Recv: warn:failed to load ./bedlevel.json: open ./bedlevel.json: no such file or directory
Recv: T:23.63 /0.00 B:23.09 /0.00 @:0 B@:0
Recv: info:config processed
Recv: T:23.63 /0.00 B:23.03 /0. T:23.61 /0.00 B:22.85 /0.00 @:0 B@:0
Recv: T:23.46 /0.00 B:22.85 /0.00 @:0 B@:0
Recv: T:23.65 /0.00 B:22.89 /0.00 @:0 B@:0
Recv: no page_ready seen, sending anyways
Recv: info:gathering device settings
Recv: info:stepd initialized
Recv: pages_ready
Recv: echo:  G21    ; Units in mm (mm)
Recv: echo:  M149 C ; Units in Celsius
Recv: echo:; Steps per unit:
Recv: echo: M92 X80.00 Y80.00 Z800.00 E95.00
Recv: info:max vel (step limit) is [2304 2304 230.4 1940.2105263157894]
Recv: echo:; Maximum feedrates (units/s):
Recv: echo:  M203 X500.00 Y500.00 Z100.00 E5000.00
Recv: echo:; Maximum Acceleration (units/s2):
Recv: echo:  M201 X500.00 Y500.00 Z100.00 E5000.00
Recv: echo:; Acceleration (units/s2): P<print_accel> R<retract_accel> T<travel_accel>
Recv: echo:  M204 P900.00 R1000.00 T900.00
Recv: info:finished gathering device settings
Recv: echo:; Advanced: B<min_segment_time_us> S<min_feedrate> T<min_travel_feedrate> X<max_x_jerk> Y<max_y_jerk> Z<max_z_jerk> E<max_e_jerk>
Recv: echo:  M205 B20000.00 S0.00 T0.00 X20.00 Y20.00 Z0.40 E5.00
Recv: echo:; Auto Bed Leveling:
Recv: echo:  M420 S0 Z10.00
Recv: echo:  G29 W I0 J0 Z-0.01550
Recv: echo:  G29 W I1 J0 Z0.00400
Recv: echo:  G29 W I2 J0 Z-0.01400
Recv: echo:  G29 W I3 J0 Z-0.00725
Recv: echo:  G29 W I4 J0 Z0.01375
Recv: echo:  G29 W I0 J1 Z0.03300
Recv: echo:  G29 W I1 J1 Z0.02625
Recv: echo:  G29 W I2 J1 Z-0.00325
Recv: echo:  G29 W I3 J1 Z0.01150
Recv: echo:  G29 W I4 J1 Z0.01200
Recv: echo:  G29 W I0 J2 Z0.03700
Recv: echo:  G29 W I1 J2 Z0.03450
Recv: echo:  G29 W I2 J2 Z-0.01575
Recv: echo:  G29 W I3 J2 Z-0.00200
Recv: echo:  G29 W I4 J2 Z0.00700
Recv: echo:  G29 W I0 J3 Z0.03700
Recv: echo:  G29 W I1 J3 Z0.03575
Recv: echo:  G29 W I2 J3 Z-0.02000
Recv: echo:  G29 W I3 J3 Z-0.00350
Recv: echo:  G29 W I4 J3 Z-0.00575
Recv: echo:  G29 W I0 J4 Z0.03075
Recv: echo:  G29 W I1 J4 Z0.00775
Recv: echo:  G29 W I2 J4 Z-0.03575
Recv: echo:  G29 W I3 J4 Z-0.02175
Recv: echo:  G29 W I4 J4 Z-0.03725
Recv: echo:; Material heatup parameters:
Recv: echo:  M145 S0 H200.00 B60.00 F0
Recv: echo:  M145 S1 H240.00 B100.00 F0
Recv: echo:; PID settings:
Recv: echo:  M301 P20.23 I1.27 D80.29
Recv: echo:  M304 P179.76 I35.67 D604.01
Recv: echo:; Power-Loss Recovery:
Recv: echo:  M413 S1
Recv: echo:; Z-Probe Offset (mm):
Recv: echo:  M851 X-45.00 Y-9.00 Z-1.30
Recv: echo:; Filament load/unload lengths:
Recv: echo:  M603 L0.00 U20.00
Recv: echo:; Filament runout sensor:
Recv: echo:  M412 S1 D7.00
Recv: Error:Line Number is not Last Line Number+1, Last Line: 3
Recv: Resend: 4
Printer requested line 4 but no sufficient history is available, can't resend
Recv: T:23.59 /0.00 B:23.01 /0.00 @:0 B@:0
Recv: T:23.52 /0.00 B:23.07 /0.00 @:0 B@:0
Recv: T:23.63 /0.00 B:22.99 /0.00 @:0 B@:0
Recv: T:23.55 /0.00 B:22.87 /0.00 @:0 B@:0
Recv: T:23.67 /0.00 B:22.93 /0.00 @:0 B@:0
Recv: T:23.57 /0.00 B:23.03 /0.00 @:0 B@:0
Recv: T:23.59 /0.00 B:23.03 /0.00 @:0 B@:0
Recv: T:23.52 /0.00 B:23.12 /0.00 @:0 B@:0
Recv: T:23.61 /0.00 B:22.97 /0.00 @:0 B@:0
Recv: T:23.65 /0.00 B:23.01 /0.00 @:0 B@:0
Recv: T:23.61 /0.00 B:22.99 /0.00 @:0 B@:0
Recv: T:23.52 /0.00 B:22.95 /0.00 @:0 B@:0
Recv: T:23.50 /0.00 B:22.91 /0.00 @:0 B@:0
Recv: T:23.63 /0.00 B:23.03 /0.00 @:0 B@:0

So its obviously connected, but Octoprint shows "Connecting..."   doesn't seem to know its actually connected