colinrgodsey/step-daemon #20: Error on connection

cinealfa, 2022-07-22:
Unable to successfully connect here is the terminal output:
Changing monitoring state from "Offline" to "Opening serial connection"
Connecting to Step Daemon
Changing monitoring state from "Opening serial connection" to "Error"
Changing monitoring state from "Error" to "Offline after error"
Unexpected error while connecting to serial port /dev/ttyACM0, baudrate 250000 from hook stepd: FileNotFoundError: '[Errno 2] No such file or directory: '/home/pi/.octoprint/data/stepd/stepd': '/home/pi/.octoprint/data/stepd/stepd'' @ comm.py:_open_serial:3836
Changing monitoring state from "Offline" to "Opening serial connection"
Connecting to Step Daemon
Changing monitoring state from "Opening serial connection" to "Error"
Changing monitoring state from "Error" to "Offline after error"
Unexpected error while connecting to serial port /dev/ttyACM0, baudrate 250000 from hook stepd: FileNotFoundError: '[Errno 2] No such file or directory: '/home/pi/.octoprint/data/stepd/stepd': '/home/pi/.octoprint/data/stepd/stepd'' @ comm.py:_open_serial:3836

PythonesqueSpam, 2024-01-24:
Same Here...

Changing monitoring state from "Offline" to "Opening serial connection"
Connecting to Step Daemon
Step Daemon is still updating.
Changing monitoring state from "Opening serial connection" to "Error"
Changing monitoring state from "Error" to "Offline after error"
Unexpected error while connecting to serial port /dev/ttyACM0, baudrate 500000 from hook stepd: Exception: 'Step Daemon is still updating.' @ comm.py:_open_serial:3916
Changing monitoring state from "Offline" to "Opening serial connection"
Connecting to Step Daemon
Changing monitoring state from "Opening serial connection" to "Error"
Changing monitoring state from "Error" to "Offline after error"
Unexpected error while connecting to serial port /dev/ttyACM0, baudrate 500000 from hook stepd: FileNotFoundError: '[Errno 2] No such file or directory: '/home/pytho/.octoprint/data/stepd/stepd'' @ comm.py:_open_serial:3916

I not that this (and other issues) haven't been sorted in over a year!!
If you're not going to fix this, at least remove your broken code from Marlin so others don't waste time trying to get your lapsed project working!!!

FrankBoesing, 2024-01-24:
> Same Here...
> 

> I not that this (and other issues) haven't been sorted in over a year!! If you're not going to fix this, at least remove your broken code from Marlin so others don't waste time trying to get your lapsed project working!!!

I agree. And if this is not maintained, it will only pollute Marlin and possibly prevent others from providing a working solution.

colinrgodsey, 2024-01-24:
The code in Marlin is a specific set of USB protocol and a G code which is not specifically tied to this software (which was produced mostly as a reference implementation): https://marlinfw.org/docs/gcode/G006.html

The first stack trace reported here is just the wrong USB serial port being provided: `/dev/ttyACM0`.

FrankBoesing, 2024-01-24:
https://github.com/colinrgodsey/step-daemon/issues/21#issuecomment-1843903683