colinrgodsey/step-daemon #23: Does direct stepping work from an SD card or is it serial only?

Reggi3, 2023-04-03:
I've been talking to a marlin user that is convinced that he's using direct stepping via an SD card without using step-daemon, I've gone through the code the best I can and it doesn't look like it's even possible for that to happen, I see nothing in marlin or any of the information on this repo that suggests that you can do direct stepping via SD.  

It looks like step-daemon is necessary for it to work at all, either via a self install on a raspberry pi etc. setup or via a plugin for octoprint.  It appears that the user has kludged using a BLOCK_BUFFER_SIZE of 256 (which means bypassing at least 2 sanity checks) and is convinced that the speed increase they are seeing is because the printer is using direct stepping via standard vanilla gcode from the SD.  I am happy to be proved wrong but I think that they might be seeing a speed increase due to the increased buffer and nothing to do with direct stepping.

Can you confirm the correct way to use direct stepping and stepd, please?  I'd really like to get to the bottom of this.

colinrgodsey, 2023-04-03:
handled this offline. but basically- it seems this user was benefiting from the increased buffer size. I do plan on getting 32-bit supported when I can, which could probably give some intensely increased resolution and speed.

Reggi3, 2023-04-03:
So does that mean that direct stepping doesn't work from the SD card and the speed increase was just down to using a larger buffer size?