colinrgodsey/step-daemon #25: could not find file "config.conf.example"

petaflot, 2023-11-16:
as stated in README.md, but I can't find this file :'-(

colinrgodsey, 2024-01-24:
looks like i renamed this at some point. I updated the readme, the provided file `config.hjson` is the one you'd want to modify if you're running stepd directly. Although its likely a better idea to use the octoprint plugin if using octoprint.

petaflot, 2024-01-24:
k thanks. I do think it would be better to name this file `config.hjson.example` cuz unless I want to fork the repo, everytime I do a pull it's going to be mean to me. same thing applies to bedlevel.json

I'm not using octoprint, but writing my own thing to send gcode to the machine : https://github.com/petaflot/GWiz

please correct me if I'm wrong.. step-daemon will actually "chop-chop" gcode and send a modify version of it to the printer, correct? if so, can you provide a few actual examples? thanks

petaflot, 2024-02-10:
ok.. so I finally got my printer to work and I most definitely think I should use step-daemon

README.md states:
```
* Baud rate should match value configured in Marlin.
* Page format should match the format configured in Marlin (defaults to SP_4x2_256).
```

questions:
* `config.hjson` has no entry for baudrate ; how should it be defined?
* what is the configuration directive for the page format?

thanks

closed 2024-02-13 by petaflot